ros-visualization/rqt_robot_plugins #79: rqt_robot_monitor: UTF8 Messages not displayed correctly

mitchellwills, 2015-01-24:
The robot monitor plugin does not appear to support using UTF8 strings. As shown below trying to send a UTF8 degree symbol (C2 B0) results in the first byte of the UTF8 sequence being treated as a seperate character.

[image]

dirk-thomas, 2015-01-24:
Where is the message being published? How is it encoded?

mitchellwills, 2015-01-24:
I am publishing a diagnostics message from libsensors_monitor. Not that the change that publishes the degree symbol is only is my personal fork (https://github.com/mitchellwills/linux_peripheral_interfaces/). It is being created in C++ using "\xC2\xB0". This should be a unicode degree symbol (http://www.utf8-chartable.de/). Using rostopic echo displays the raw diagnostics message correctly. The ROS message wiki page (http://wiki.ros.org/msg) also encourages using UTF8 for encoding unicode strings.

dirk-thomas, 2015-01-26:
The Python code of this plugin does not perform any decoding as far as I can see:

* https://github.com/ros-visualization/rqt_robot_plugins/blob/hydro-devel/rqt_robot_monitor/src/rqt_robot_monitor/status_item.py#L67
* https://github.com/ros-visualization/rqt_robot_plugins/blob/hydro-devel/rqt_robot_monitor/src/rqt_robot_monitor/status_snapshot.py#L64

Please consider to provide a pull request adding decoding of the message fields.

dirk-thomas, 2017-04-25:
This issue was moved to ros-visualization/rqt_robot_monitor#2

closed 2017-04-25 by dirk-thomas