ros-visualization/rqt_robot_plugins #81: rqt_rviz: Listening for Polygon causes segfault and crash

alexbsher, 2015-01-30:
Using rqt_gui with a number of plugins including rqt_rviz, trying to visualize a polygon in rqt_rviz causes the whole gui to crash. We're using Indigo on Ubuntu 14.04. The same topic is viewable in rviz alone, just not in rqt_rviz.

dirk-thomas, 2015-01-30:
Can you please describe a specific use case in order to reproducible the issue.

alexbsher, 2015-01-30:
We're using Polygons for our robot_footprint topic. When adding the view inside of rqt_rviz the gui crashes. Is there anything specific I can post? A sample message? The configuration of our rqt_gui?

dirk-thomas, 2015-01-30:
Any kind of instructions to reproduce the problem without involving guessing would be very helpful.

E.g. a simple script publishing the specific message and instructions what you do to visualize them which leads to the crash.

alexbsher, 2015-01-30:
Ok see if this works without getting into too much configuration:

1) roscore
2) rosrun rqt_gui rqt_gui
3) Add an RViz plugin to the rqt_gui
4) Run:
rostopic pub -r 10 /test_poly geometry_msgs/PolygonStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: '/map'
polygon:
  points:
  - x: 0.0
    y: 0.0
    z: 0.0" 
5) Try to add a display for that /test_poly topic (either by topic or type)

Those steps cause the entire rqt_gui to crash with our setup.

dirk-thomas, 2015-01-30:
I can reproduce the problem (which doesn't happen in standalone rviz). I assume this is a duplicate of ros-visualization/rqt#96.

You could try an older version of `rqt` as described in the referenced ticket to verify that it is the same problem.

alexbsher, 2015-01-30:
Followed the steps in the other ticket, and this seems to be the same problem, except the only kind of topic that triggers the segfaults are messages of type geometry_msgs/PolygonStamped. Are there plans ot fix this in the supported versions of rqt_rviz?

dirk-thomas, 2015-01-30:
As mentioned in the referenced ticket I don't think this is fixable in rqt itself. The rviz API does currently not provide a way to pass in an external spinner. I assume this requires modifications to rviz to be addressed (cc @wjwwood).

samuelba, 2015-09-17:
the problem is solved with this https://github.com/ros-visualization/rviz/pull/831 pull request, however nobody seems to want to merge it.

ablasdel, 2015-09-21:
It seems there are unaddressed concerns on PR are you looking into fixing those?

dirk-thomas, 2016-09-08:
I will close this ticket as a duplicate of the reference `rviz` ticket.

closed 2016-09-08 by dirk-thomas