ros-visualization/rqt_robot_plugins #85: Improper Rendering of text in rqt applications

sidcode, 2015-05-30:
Configuration - 
ROS Indigo Version 1.11.13
Ubuntu 14.04
rqt_tf_tree 0.4.1
rqt_graph 0.3.11

The text seems to be filling out of the boxes representing nodes and services. 
The attached image showing the issue -
[image]

Note that this is not a rqt_tf_tree specific problem. It persists across all rqt applications for me.

unusual-thoughts, 2015-11-25:
I have the same issue
Config:
Arch Linux
ros-jade-desktop 1.2.0
rqt-tf-tree 0.4.2
rqt-graph 0.3.12
3200x1800 screen

chapulina, 2015-12-09:
This could be an issue with the [box shape](https://github.com/ros-visualization/qt_gui_core/blob/groovy-devel/qt_dotgraph/src/qt_dotgraph/node_item.py#L53) in `qt_dotgraph`

chapulina, 2015-12-10:
I traced the issue back to [this](https://github.com/ros-visualization/qt_gui_core/blob/groovy-devel/qt_dotgraph/src/qt_dotgraph/pydotfactory.py#L163) line in  `ros-visualization/qt_gui_core/qt_dotgraph` (`dot = graph.create_dot()`). 

The input graph contains no information about node width, but the output dot file does. If you have a box node and an ellipse node with the same text, the box node will receive a smaller width and eventually a smaller bounding box. The larger the text, the more visible the problem is.

Since it seems like an upstream issue, a temporary solution could be to multiply the box node's bounding box by a factor, and translate it accordingly, before creating it [here](https://github.com/ros-visualization/qt_gui_core/blob/groovy-devel/qt_dotgraph/src/qt_dotgraph/node_item.py#L54).

DorianScholz, 2016-11-27:
I have committed a fix in https://github.com/ros-visualization/qt_gui_core/pull/73
It would be greatly appreciated, If you could test it and give feedback.

unusual-thoughts, 2016-12-15:
The fix works perfectly (merged on kinetic in [unusual-thoughts/qt_gui_core](https://github.com/unusual-thoughts/qt_gui_core))

dirk-thomas, 2017-01-24:
I assume this has been addressed by ros-visualization/qt_gui_core#73.

closed 2017-01-24 by dirk-thomas