ros-visualization/rqt_robot_plugins #87: Probably interesting additions rqt_batteries and rqt_emergency_buttons

awesomebytes, 2015-06-17:
Hello, as part of an effort on having some ready-to-use (i.e.: no coding involved) rqt status visualization of robots I've created:

https://github.com/pal-robotics/rqt_batteries
https://github.com/pal-robotics/rqt_emergency_buttons

Which, provided you set some params in the param server like:

```rosparam set /batteries_dashboard/batteries "{'battery1': {'percentage_topic': '/percentage1', 'charging_topic': '/charging1', 'tooltip_name': 'BATT1'}}"```

It will give you a little rqt widget (very tiny!) with just the battery status or the emergency button status respectively.

I wonder if it would make sense to merge this into this repository as it's a general tool.

In any case you may be interested in adding the file:
https://github.com/pal-robotics/rqt_emergency_buttons/blob/master/src/rqt_emergency_buttons/icon_widget.py

Which let's you create a generic icon to show a state of something with no behavior when clicking (used to show the emergency button status in this case). 

The widgets can also be used in the way the rest of the rqt_robot_plugins are presented.

ablasdel, 2015-06-17:
I believe it would make a good deal of sense to add this in as a default. To do so of course you would have to let them be BSD licensed when they are committed here. Is that OK? 

The main reason I didn't make a "default robot dashboard" when I made this framework is because there wasn't enough standardization of what a robot would want in their dashboard so I made it a "code your own from these pieces".

Would you be interested in making something generic for "all robots" like that? Either way I think we should get this work merged into the main line if possible. I would LOVE to have a "general all robot" rqt_dashboard that is really useful for almost all robots and then modifiable with new widgets if needed.

awesomebytes, 2015-06-17:
As those are deeply inspired in this code they are already BSD if you look in the package.xml.

In the works of making a "default robot dashboard" I made this little pieces that can be invoked just by setting some ros params. I don't know if its the best way to go for this problem, but I think it does the trick. In this case using launch files does not work, and putting a configuration window on it was... kinda painful. I'm also working in a text field widget, which you give a topic (or subtopic) and it shows it's current content there, by my experience in robocup rescue it's quite useful to be able to add some debugging data, typically a `rostopic echo`, in the rqt dashboard of your robot (and adding a bash rqt plugin and write in it the command every time you open the rqt again is quite tiresome).

In even more early status I've got a button-to-publish on a topic and button-to-do-a-service-call like widgets which also make prototyping a GUI for making a robot do something pretty easy. Once again coding the less as possible, just setting params. I've been commented before that there is a topic and service plugin to do this, but it opens a very big window and when you just want to have a single button to do that job, it's an overkill.

ablasdel, 2015-06-17:
Sorry I was looking at some of the files and they are stamped with a different license at the top. If BSD is intended then that is great! I'll look at how we can best fit this with robot_dash to make a general dashboard soon.

dirk-thomas, 2017-04-25:
All plugins have been moved into separate repositories (#123). Please consider to create your own repositories for new plugins. They can be added to the metapackage in this repo on request.

closed 2017-04-25 by dirk-thomas