ros-visualization/rqt_robot_plugins #88: Release for indigo

ablasdel, 2015-07-23:
- [ ] Need to make indigo-devel branch
- [ ] Need to keep hydro-devel branch
- [ ] Need to roll back https://github.com/ros-visualization/rqt_robot_plugins/commit/06e51d1f5b4ed9c926f28779054ba143a016bcc9 in hydro-devel since 3 as stale does not exist in hydro.
- [ ] Re-release for hydro, indigo & groovy

dirk-thomas, 2015-07-23:
Why is that necessary? While in hydro the level can never be 3 the patch does not harm on hydro.

ablasdel, 2015-07-23:
Yeah you are right it will never be 3 so i guess it is innocuous. Still want to release and make an indigo-devel branch.

ablasdel, 2015-07-23:
My plan is change the branch name for each LTS. Does that sound OK?

That said maybe we should steer away from this and just go back to "master" which is used on everything unless there is a specific conflict and then we can spin off a backports branch? That way the branch name never gets stale.

dirk-thomas, 2015-07-23:
That is okay with me. You need to update all references in the distribution files as well the gbp repo for that.

Also keep in mind that such a change requires everybody who has a checkout to switch to the new branch.

Btw. there will certainly not be a groovy release anymore since groovy packages are not being built anymore. And for hydro you should be dead certain that the new release doesn't break anything because you might not get another shot to release into hydro once the build farm is retired (which can happen anytime after it is EOLes which is already the case).

ablasdel, 2015-07-23:
Hydro doesn't need a release then. that commit doesn't actually change anything on hydro

ablasdel, 2015-07-23:
Just planning to do indigo & jade then

dirk-thomas, 2015-07-23:
Sounds good to me.

ablasdel, 2015-07-24:
Done and done

closed 2015-07-24 by ablasdel