ros-visualization/rqt_robot_plugins #95: rqt_rviz: No way to disable or avoid creation of rqt_rviz_ogre.log file in working directory

pallgeuer, 2016-01-05:
In file rqt_rviz/src/rqt_rviz/rviz.cpp, in RViz::initPlugin(qt_gui_cpp::PluginContext& context), an Ogre log manager is created, followed unavoidably by a rqt_rviz_ogre.log file. When I start rqt (which I do from varying directories on the command line) the effect of this is that my directory tree is being polluted by rqt_rviz_ogre.log files, which I then have to go through and clean out for example when I want to commit in a code repository.

A way of avoiding the creation of this file (or disabled by default and a way of enabling it) would be useful. I use rqt heavily every day, but I've never in all these years had a reason to look inside the ogre log file.

dirk-thomas, 2016-01-05:
This plugin can only configure `rviz` using its API and afaik currently `rviz` doesn't provide an API to disable the Ogre log manager (@wjwwood Please correct me if I am wrong).

@pallgeuer Therefore I think you should fill this as an enhancement request against rviz (or even better provide a PR to add the option to the API). Then this plugin can make use of the new API and disable the Ogre log manager by default.

wjwwood, 2016-01-07:
You can apparently control that from the rviz command line: https://github.com/ros-visualization/rviz/blob/428126c7f59ccc732f3e92f8ce54439e05d95805/src/rviz/visualizer_app.cpp#L134

Not sure if it works or if it is accessible from the rviz API.

wjwwood, 2016-01-07:
Maybe you can use this API? I don't know how rqt is wrapping rviz: https://github.com/ros-visualization/rviz/blob/indigo-devel/src/rviz/ogre_helpers/ogre_logging.h#L56

pallgeuer, 2016-01-07:
Not that I have a good overview of rviz/rqt and what is done where, but in my eyes the log file is created here:
https://github.com/ros-visualization/rqt_robot_plugins/blob/master/rqt_rviz/src/rqt_rviz/rviz.cpp#L78
This is inside the code that wraps rviz into rqt. If I just start rviz directly I don't get a log file.
So I don't think trying to change something inside rviz using the rviz API is the solution here. If it were to be disabled by default in that function, but with the option to enable it via the arguments that are parsed immediately prior, then that would do it I think.

dirk-thomas, 2016-01-07:
I have created PR #96 which disables the logging by default as well as adds a command line option to enable the ogre log. This should mimic the behavior of rviz better and by default don't pollute the current folder.

It would be great if you could try the branch and confirm that it fixes the problem for you.

pallgeuer, 2016-01-07:
Thanks! I tested it out and it works for me.

dirk-thomas, 2016-01-07:
Thanks for checking!

commit 08322e1c (2016-01-07): disable ogre-log by default, add command line option to enable it (fix #95)
closed 2016-01-07 by dirk-thomas