ros-visualization/rqt_robot_plugins #103: Runtime_Monitor_Widget AttributeError

anderwm, 2016-05-09:
On Kinetic (probably more because of my qt version, 5.5.1) Runtime_Monitor_Widget crashes rqt_gui with
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_runtime_monitor/runtime_monitor_widget.py", line 174, in _update_messages
    if self.tree_widget.isItemSelected(item.tree_node):
AttributeError: 'QTreeWidget' object has no attribute 'isItemSelected'

Python is correct, QTreeWidget::isItemSelected is deprecated according to [QT](http://doc.qt.io/qt-4.8/qtreewidget-obsolete.html#isItemSelected) in favor of QTreeWidgetItem::isSelected()

dirk-thomas, 2016-05-10:
Can you please try if #104 fixes the problem for you.

anderwm, 2016-05-10:
Yep, thanks

dirk-thomas, 2016-05-10:
Thank you for trying it.

closed 2016-05-10 by dirk-thomas