ros-visualization/rqt_robot_plugins #108: problem with MonitorDashWidget addition to dashboard

twiniars, 2016-06-01:
in ros kinetic addition of 
**self._monitor = MonitorDashWidget(self.context)**
to dashboard causes the following, cyclic (ca. 1s) console output:
**QObject::startTimer: Timers cannot be started from another thread
QObject::killTimer: Timers cannot be stopped from another thread**

dirk-thomas, 2016-06-01:
Can you please describe what exactly you ran and what the output was. From your description I don't see which line you are referring to.

twiniars, 2016-06-01:
I run the python script in the attachment. The scripts runs the QT application with the MonitorDashWidget. As a result the QT prints an error on the console.

QObject::startTimer: Timers cannot be started from another thread
QObject::killTimer: Timers cannot be stopped from another thread

[irp6_dashboard.py.txt](https://github.com/ros-visualization/rqt_robot_plugins/files/293728/irp6_dashboard.py.txt)

dirk-thomas, 2016-06-01:
Have you tried to use the class according to its [documentation](https://github.com/ros-visualization/rqt_robot_plugins/blob/8c416b0a34aa944f3ab8f010e709e0c374bffa2d/rqt_robot_dashboard/src/rqt_robot_dashboard/widgets.py)?

twiniars, 2016-06-01:
When I use the code almost directly from the documentation you mentioned, the problem still occurs. I attach the file. Please note the mistake in doc. The line 
`from rqt_robot_dashboard.widgets import MonitorDashWidget, ConsoleDashWidget`
should be replaced with
`from rqt_robot_dashboard.widgets import MonitorDashWidget, ConsoleDashWidget, BatteryDashWidget`

If I do something wrong please let me know. 

[irp6_dashboard.py.txt](https://github.com/ros-visualization/rqt_robot_plugins/files/294428/irp6_dashboard.py.txt)

dirk-thomas, 2016-06-01:
Maybe @ablasdel can comment on how he is using the dashboard in custom rqt plugins.

twiniars, 2016-06-01:
Fortunately, from my point of view the problem is not critical.

twiniars, 2016-06-01:
I solved the problem with signals and slots according to 
http://pyqt.sourceforge.net/Docs/PyQt5/signals_slots.html. See pull request #106

closed 2016-09-08 by dirk-thomas