ros-visualization/rqt_robot_plugins #111: Unable to install from source

chrisspen, 2016-09-13:
I attempted to install this package from source on a Raspberry Pi using the following procedure:

```
cd /home/pi/ros_catkin_ws
rosinstall_generator rqt_robot_monitor --rosdistro kinetic --deps | wstool merge -t src -
wstool update -t src -j2 --delete-changed-uris
rosdep install --from-paths src --ignore-src --rosdistro kinetic -y -r --os=debian:jessie
./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release --install-space /opt/ros/kinetic -j1
```

No error was reported. However, `rospack list | grep -i monitor` shows nothing was installed.

Your package summary lists no install documentation, so it's unclear if this is a valid way to install. Is this unsupported, or did I make a mistake in my procedure?

dirk-thomas, 2016-09-13:
Please ask your question on answers.ros.org according to the [support guidelines](http://wiki.ros.org/Support#Guidelines_for_asking_a_question_.28Please_read_before_posting.29).

closed 2016-09-13 by dirk-thomas

chrisspen, 2016-09-15:
That was an incredibly rude and dismissive reply. For the record, [I did ask there](http://answers.ros.org/question/243625/installing-rqt_robot_monitor-on-kinetic/), but people rarely answer.

And like I said, this isn't a problem with me not knowing how to install packages from source. I'm running on a Raspberry Pi, which doesn't have any pre-compiled ROS packages, so I've installed every Kinetic ROS package from source. Yours is the only one that's failing, suggesting it's a problem with your package.

dirk-thomas, 2016-09-15:
I apologize if you perceived the reply as being rude. That was not my intention.

We do receive a very large amount of tickets and questions like this should be asked on answers.ros.org. The reason for that are the following:
* There are many more people reading questions there so you chances on getting an answer and in a timely manner are much higher.
* In the future others users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such a action item.

Regarding your statement "people rarely answer": the statistics says that 75% of the questions have accepted answers. So the rate is pretty good. Especially if you consider that many users write answers in there free time and are not being payed to do so.

You also didn't mention that you asked the question already, e.g. by providing a link to the question. But on the other hand you posted both the question on answers.ros.org and this ticket very close to each other. In my opinion it is ok to post a question in the issue tracker after it got no attention on answers.ros.org for several days but that is not the case here. Cross posting the same question through several channels is also something which the Etiquette in the support guidelines explicitly mentions.