ros-visualization/rqt_robot_plugins #115: Add gauges

AndyZelenak, 2016-11-10:
My research group was interested in displaying virtual gauges, so I wrote a standalone package for it. 

https://github.com/UTNuclearRoboticsPublic/gauges

It's very simple: you just specify the topic name, namespace, and range for each gauge in a launch file. A drawback is that it pops up individual windows. How hard would it be to add integrate that here?

Image: https://postimg.org/image/s14ugru5f/

dirk-thomas, 2016-11-10:
Afaik there are already plenty of gauge widgets for Qt out there. Newer Qt versions even have a QML type for that (http://doc.qt.io/qt-5/qml-qtquick-controls-styles-circulargaugestyle.html). So I would think you don't need to implement your own widget.

Regarding making this a plugin that should be pretty straight forward. The [wiki page of rqt](http://wiki.ros.org/rqt) contains tutorials how to write your own plugins. If you don't have a strong need for C++ I would recommend using Python instead. Its easier to write, maintain, and use from my experience.

--

I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2016-11-10 by dirk-thomas

AndyZelenak, 2016-11-10:
Thanks for the insightful response. I would have called it a "feature request." So if I write a Python gauge plug-in, a pull request would likely be accepted?

(I don't want to waste my time, otherwise)

dirk-thomas, 2016-11-10:
In general we suggest people to keep the plugin in their own repo and release it independently. There are already "too many" plugin in this single place. And for maintainability it is easier to keep them separate so that every maintainer can release following his own schedule.

AndyZelenak, 2016-11-30:
Well, it's done. I wouldn't call it straightforward, though  ; )  especially compared to the MATLAB gui's I'm used to.
https://github.com/UTNuclearRoboticsPublic/gauges/tree/master/rqt_gauges